mcer12/Hugo-ESP8266 #6: Battery reporting topics

giddyhup, 2019-08-09:
Great soft- and hardware! Apparently, a Hugo reports its battery level under two topics, is this on purpose? Also, 0x3e is 62 not 90, what does it stand for? 

```
esp/hugh/battery 3e
esp/hugo/battery 90

```

mcer12, 2019-08-13:
Sorry for getting back to you so late, I didn't receive a notification... this is a bug and will be solved in the upcoming update. I'll comment here once its pushed in the git.

mcer12, 2019-08-17:
@giddyhup This should also be fixed with the latest sketch.

giddyhup, 2019-08-17:
Thanks for the update. The line `publishBatteryLevel();` is missing in the code for button 1.

mcer12, 2019-08-17:
Thanks, you're correct. I pushed your solution.
I will add an option for custom topic in the next update.

giddyhup, 2019-08-18:
Good idea. In my case I named my two Hugos (red and black) and put the respective names in the topics.

mcer12, 2019-08-24:
@giddyhup Check out the latest update, it's a bigger one.

giddyhup, 2019-08-24:
Thanks. I will check it out when I return from my trip in a couple of weeks. Meanwhile, I'll close this issue.

closed 2019-08-24 by giddyhup