mcer12/Hugo-ESP8266 #8: Hugo_MqttTrigger Access Point connection error

dimosap, 2019-08-26:
Hello, 

I am a fan of your work and i'm testing your Hugo_MqttTrigger code on an esp12e module i own. Unfortunately i have the following problem. The access point appears only for some seconds and i'm not able to connect to it. After that it disappears from the available networks. Any ideas on the cause of this?

Thank you in advance.

mcer12, 2019-08-26:
Hi, this is very unusual and I didn't encounter anything like that. Few things to check off the top of my head.
1. Did you erase flash (all flash contents) when flashing the esp? 
2. Are u using sufficient and stable power supply? 
3. Is it possible the device you're using to connect to the AP is at fault? Try different one.
4. Do you have a nodemcu/wemos devboard at hand to try the sketch on? Just to check if the issue is not with the module.

I don't know the hookup scheme you have so it's really difficult to pinpoint what might be the issue. Try the above suggestions and if that doesn't help, try to provide me more details on your setup.

mcer12, 2019-08-26:
Oh and it can be caused by the button being still pushed down, for whatever reason. Maybe you didn't hook up the buttons at all and they are floating, maybe you have too large capacitor and it keeps the button pushed for too long. After 3 seconds after starting the config AP, program checks the buttons and if any of them is pushed, AP shuts down.

mcer12, 2019-08-26:
Did you check out my second post too? I think it's very likely to be the issue. Maybe try the buttons without the capacitor and make sure the pulldowns are in place. You will have to hold a button for half a second for it to get recognised but if my theory is correct, the AP won't disappear. This would mean the capacitance is too high, also esr can come into play here so use ceramics.

dimosap, 2019-08-26:
Thanks for your fast response, 
I program the esp12e through an FTDI. I don't use other parts but only an ams1117 and buttons. I've also tested it with a nodemcu and a different device to test the connection but i came across the same results.

dimosap, 2019-08-26:
It finally worked!!!
The problem was that I had not connected all the buttons. Now I need to figure out where to put the link from ifttt trigger in the Hugo Configuration.
Thank you a lot for your help, your work is amazing!

mcer12, 2019-08-27:
Glad I could help!

closed 2019-08-27 by mcer12