mcer12/Hugo-ESP8266 #9: MQTT TLS support

juzam, 2019-08-27:
First of all, thanks for this nifty little gadget, I really like mine. I'm using it with the MQTT firmware 1.3b which added username and password support.

My only desiderata at this point would be support to connect to a secure MQTT broker (with optional client verification). 

Would it be hard to implement this? more importantly, would it fit in the maximum space/memory constraints?

Thank you again for your time and effort spent on the project.

mcer12, 2019-08-27:
Hi @juzam I don't currently have the means to do this, I don't use tls at home, I have separate wifi network for IoT so I don't really have a reason to. But it shouldn't be too difficult to implement, it should fit since current sketch uses only around 40% memory but OTA will probably not work since it needs 50% free space. Check this out:
https://github.com/knolleary/pubsubclient/issues/462
Of course any contribution is appretiated!

juzam, 2019-08-27:
Thanks for providing that link, I think I can do some tests with that. I'll report here my findings.

mcer12, 2019-10-10:
Hi @juzam have you made any progress with the TLS?

juzam, 2019-10-10:
I'm so sorry but I didn't have any spare time to work on TLS support.  What I've seen from the link you posted is that `WifiClientSecure` could be an almost drop-in replacement for `WifiClient`. I hope to get some spare time during the weekend to do some tests in that regard.

juzam, 2019-10-21:
I had some time to work on this during the weekend but I still don't have anything usable. I've also realized that having just one remote doing the compile/upload/reconfigure/test cycle is very time consuming. I'm waiting for an esp8285 to be delivered to ease the process. 

meanwhile I've setup up a fresh mosquitto broker with let's encrypt certificates to start with a clean slate.

mcer12, 2019-10-21:
@juzam you can use any ESP8266 module or devboard, it's fully compatible with esp8285 ;)

juzam, 2019-12-16:
a little update. I'm stuck at not being able to connect to a tls secured broker. I've tried a simple apporach with a very stripped down arduino sketch to no avail. What I'll do next is take a closer look on how `Tasmota` deals with mqtt and tls. They are using ``` #include <t_bearssl.h>``` to do so.

also, apologies for my perennial lack of time to work on this.

Darkmelman, 2020-08-13:
@juzam and @mcer12 Some time ago I implemented a very similar project myself with an Adafruit Huzzah32 and tls support. I think you could take over the parts of the source code that concern tls and WifiClientSecure

[https://github.com/Darkmelman/mqtt_button.git](url)

As soon as I have the time, I will be happy to support you

mcer12, 2020-08-13:
Hopefully @juzam will be able to use that, thank you for the hint! Though I don't use TLS in mqtt and don't have time to dig into it, I will accept PR if you decide to contribute! 

I see you use adafruit mqtt library, did you have any particular reason for that? I ask because this would probably not fit within ESP-M2 memory together with pubsub client and it would have to be made as a separate sketch ;) I had to optimize quite a bit to squeeze HTTP + HTTPS capability in url trigger sketch.