mcer12/Hugo-ESP8266 #10: Documentation regarding config mode is hard to find.

jtokash, 2019-10-19:
Please put a link to https://github.com/mcer12/Hugo-ESP8266/wiki/Sketch:-HTTP-(Basic-URL-trigger) in the main page of the wiki.

mcer12, 2019-10-19:
Hi @jtokash I don't quite get what you mean. All wiki sections are on the main page of the wiki as well as on project's home page. What changes do you propose?

jtokash, 2019-10-19:
Sorry for the confusion.  I clicked through each of the links and didn't get what I needed,  so I didn't realize you were trying to link to it.  Turns out the link you have is just typo'd.

On the main page of the wiki, there is a link to  Sketch:-HTTP-(Basic-URL), which does not exist.  The page you want to link to is Sketch:-HTTP-(Basic-URL-trigger).

mcer12, 2019-10-19:
Oh, I see! Sorry for that and thanks for pointing that out. I renamed the section and forgot to change the links... Fixed now!

closed 2019-10-19 by mcer12