mcer12/Hugo-ESP8266 #11: RFE : include printed MAC address on devices, web config interface

MagicFab, 2019-10-27:
It would be very useful to know the MAC address in order to be able to assign a static IP address before the first connection.
I'd like to suggest including the MAC address of each device in printed documentation or sticker on the device/static bag it comes in. This should also be displayed in the Hugo web configuration.

mcer12, 2019-10-27:
Hi @MagicFab at this point, the MAC address is appended to OTA name, I think the easiest approach is to append lets say last 6 characters to config SSID. Something like HugoXXXXXX instead of HugoConfig. 
I will close the issue and let you know once it's done.

MagicFab, 2019-10-28:
Sorry I wasn't clear, I was not referring to the OTA name but to actually showing it in the web UI at http://10.10.10.1.

mcer12, 2019-10-28:
Yes this is a good idea, I will add it somewhere in the config portal.

mcer12, 2019-10-30:
MAC address was added in config portal as well as in config portal SSID, check out latest fw!

closed 2019-10-30 by mcer12

MagicFab, 2019-10-31:
Thank you! I've upgraded the firmware on one of the remotes I have and the MAC address is now show, most excellent!