mcer12/Hugo-ESP8266 #12: RFE : Make "HugoConfig" AP name editable, unique by appending last 4 characters of MAC address

MagicFab, 2019-10-27:
I have several Hugo remotes and they all advertise SSID "HugoConfig" when in AP mode.
It would be very useful to see them as "HugoConfig-XXXX" for example to distinguish them.
Furthermore it would be nice to be able to rename their SSID to whatever we want, once initially configured.

mcer12, 2019-10-27:
I think the previous issue will resolve this one as well. I will change the config portal SSID to something like HugoXXXXXX (last 6 MAC address characters).
I will not add an option to change the SSID as I want to keep the options portal as simple as possible. That being said, you can edit the sketch and change the name to your liking :)

MagicFab, 2019-10-28:
No, the previous issue is different (see my comment there).
I noticed right now the SSID is ESP_XXXXXX, I am not sure this wasn't coming up for me. Yes, having it as HUGO_XXXXXX would still be more intuitive and coherent.
The interface can be kept simple if the editable field shows the current SSID while making it editable, that way it would provide useful information while making it editable.

mcer12, 2019-10-30:
Check out latest fw, it should address your request with the exception of SSID name editability. SSID name will not be editable by default but you can edit it directly in the sketch if you want to.

closed 2019-10-30 by mcer12