mcer12/Hugo-ESP8266 #13: Battery topic is not included in the discovery (doHassRegister() ) function

mikeage, 2019-10-27:
Calling doHassRegister() sends a config topic for the state, but doesn't include battery information. 

I'll try to submit a PR once I get an environment for flashing set up.

mikeage, 2019-10-27:
I'm having trouble flashing a new version (see #14 ), but here's a discovery string sample for battery (assuming the default topics); this one should be posted to `homeassistant/sensor/hugo_bd2916/battery/config`
```
{"unit_of_measurement":"%","device_class":"battery","state_topic":"homeassistant/sensor/hugo_bd2916/battery","name":"Hugo bd2916 Battery","unique_id":"hugo_battery_bd2916","device":{"ids":"hugobd2916","name":"Hugo - WiFi remote","mf":"https://github.com/mcer12/Hugo-ESP8266", "mdl":"Hugo-ESP8266","sw":"1.3"}}'
```
This is in conjunction with a regular discovery string of:
```
{"unique_id":"hugo_bd2916","name":"Hugo bd2916","stat_t":"homeassistant/sensor/hugo_bd2916/state","ic":"mdi:remote","exp_aft":"1","device":{"ids":"hugobd2916","name":"Hugo - WiFi remote","mf":"https://github.com/mcer12/Hugo-ESP8266", "mdl":"Hugo-ESP8266","sw":"1.3"}}
```
The device ID is the same, so they're linked in HA, but there's a unique `unique_id` for the battery sensor.

mcer12, 2019-10-27:
Hi @mikeage yes, currently battery state is not included in the discovery, this is one of the priorities for the next release. It should be ready sometime during next week, will keep you posted.

mikeage, 2019-10-27:
Ok, cool. Figured  I’d try to help a bit, but I’m not yet at the stage to submit a PR...

mcer12, 2019-11-03:
Check out latest commit, all you have to do is register Hugo again for it to update the configuration. Your suggested payload was a great help.

closed 2019-11-05 by mcer12