mcer12/Hugo-ESP8266 #14: Trouble flashing OTA

mikeage, 2019-10-27:
Hi,
I realize this might be a generic Arduino question; most of my experience is with platformio, and I'm not completely comfortable in the Arduino GUI.

I've made a small change to the code, compiled it, put the Hugo into OTA mode by pressing buttons 1+3 (rapid flashing), selected the network port, and selected Upload in the GUI. It says it's uploading, the LED becomes solidly lit, and then the Arduino GUI reports upload complete, but the LED stays lit and it seems like it did not accept the new .bin. If I push a button, any button, it seems to restart, but it does not appear to have loaded the new SW (tested by updating `#define FW_VERSION` and pressing buttons 2+4 to send the config string.

Any idea what I might be missing?

mcer12, 2019-10-27:
After the OTA finishes, it can still be processing the binary for few seconds. Just to be sure, by "not appear to have loaded the new SW" you mean the device is unresponsive or just acts like nothing happend?

mikeage, 2019-10-27:
I mean it stays with the LED lit for a while. If I press a button, it sends MQTT messages again, but with the old FW VERSION string.

mcer12, 2019-10-27:
Can you try to flash it again with the OTA? It might've been some kind of unusual hickup, usually if the OTA fails the device crashes. Also make sure to double chceck if all the options correspond with wiki page, if the settings were set wrong, binary might've been refused by the device:
https://github.com/mcer12/Hugo-ESP8266/wiki/Flashing-&-FW-update

Let me know how this goes, hopefully all is OK. Keep in mind though that OTA can fail from time to time and brick poor Hugo and since you have a version without CH340, this can get annoying :)

mikeage, 2019-10-27:
Sure (but probably not until tomorrow). I’ll let you know.

mcer12, 2019-11-03:
In case you didn't have time to try it out yet, check out latest commit, might interest you.

mikeage, 2019-11-03:
Thanks, I saw! I got pulled out of town on short notice and am now 2000km away from my devices :)

closed 2020-07-20 by mcer12