mcer12/Hugo-ESP8266 #15: Trouble registreting Hugo to diyHUE/Hue app  - docs refers to "Sensors" section but there isn't any in the Hue app

MagicFab, 2019-10-28:
The documentation for pairing a Hugo remote with diyHue indicates:

> Push and hold buttons 2 + 4 until the led lights up, and release. This will register the remote in diyHue and it will pop up in Hue app, in Sensors section.

When I push nad hold buttons 2 + 4 (assuming button "2" is under "O" and "4" is "I"), the led starts flashing fast. Is this indicating a registration attempt ?

The diyHue IP is properly configured in the Hugo remote. I was able to import lights into it from an existing Philips Hue bridge. I am not sure what is meant by "This will register the remote in diyHue", it doesn't show up anywhere I can see in the web UI. Can you provide more details ?

Regarding the Pillips Hue app, there isn't a "Sensors" section, only "Your rooms" in the Home section and "Accessory setup" under Settings.

mcer12, 2019-10-28:
Hi @MagicFab Do you use diyhue emulator or original hue bridge? Not sure how you come up with the numbering but the buttons are numbered from top to bottom so "I" is button 1, "O" is button 4 :) I will add this to wiki.

Yes Accessory is the correct menu. There might've been a change in Hue app, it was Sensors before.

MagicFab, 2019-10-28:
OK, thanks for clarifying the buttons. Perhaps the initial config/wifi connection can be a common section in the main documentation page where there is a picture, and identify the buttons on the picture. I am attaching such a miodified image, it's a GIMP XCF document with layers which you can modify (zipped, apparently Github doesn't like XCF) and a rendered jpeg image.

[image]

[hugo-buttons.zip](https://github.com/mcer12/Hugo-ESP8266/files/3779265/hugo-buttons.zip)

I am using diyHue as I don't expect the remote to register and show up on the other which is a the Philips bridge hardware device. In the Hue app both bridges are registered but I am connected to the diyHUe.

mcer12, 2019-10-30:
Double check if you entered the correct bridge IP and you are on the same network. Then push and hold buttons 2 and 4 until the led lights up for a second. Then it should pop up in the Hue app. Let me know how it goes.

mcer12, 2019-10-30:
I personally use diyhue at home so I've done this for all my remotes without issues. I would also check if the diyHue bridge is reachable using the IP address.

MagicFab, 2019-10-31:
Ok, so I had some time to check this again today.

After upgrading firmware for 2 of my devices and connecting them I tried the above with Philips Hue app and Hue essentials on one device. The remote appeared in both apps in the diyHUE, but not in its web interface (I don't think the UI shows any remotes)! 

It appeared as a Hue Tap Switch under Settings > Accessory Setup. It's not ideal as it is a round-shaped remote, buttons 1,2,3 and 4 matched the same on Hugo and I was able to program them.

Another interesting fact, once I was finished I picked the other remote to set it up and when I touched button 1 it turned on the light... same as the other remote. The second remote is programmed the same as the first, yet it wa not registered to Hue. It seems the app thinks both remotes are the same (mirrored?). 

I'll file a separate bug report, this one should be closed now. Thank you!

mcer12, 2019-10-31:
That would mean the remotes have the same MAC address which seems extremely unlikely. Maybe since you don't have it registered, diyHue doesn't know what to do and uses one of the already registered remotes. Try to register the second remote if you didn't do so already, that should solve it.

MagicFab, 2019-10-31:
OK, I'll be able to try that tonight - thanks!

mcer12, 2019-11-05:
Have you had time to try it out? If it's working as intended, please close the issue.

MagicFab, 2019-11-05:
Yes, I've tried this again after removing the registered remote and was unable to register either one of the remotes I have. I can't register it again, despite completely resetting both Hue bridge and diyHue bridge configurations completely and reflashing the devices. This worked once, never again.
What is the recommended way to reset the remote ? Is re-flashing enough ?

mcer12, 2019-11-05:
Oh I get it now! You don't have to reset the remote at all, this is diyHue issue with any remote (not just Hugo). Once you remove a remote from the app, it is still present in the diyHue config file. You have to manually remove the rmotes from diyHue file to *really* remove them and be able to register them again.  Wonder why this is still not fixed, I will add an issue on diyHue github page.

mcer12, 2019-12-16:
@MagicFab Latest diyHue version shoudl fix it but you will still have to remove the remotes from the config file or reset the diyHue to factory config.
https://github.com/diyhue/diyHue/commit/2c488808528924cd427f18bc9d473a76abd349b1

closed 2019-12-16 by mcer12

MagicFab, 2019-12-16:
OK, I never thought it would be a diyHue problem! I will be trying this again shortly - thank you!