mcer12/Hugo-ESP8266 #16: Small mistake in diyHUE pairing documentation

MagicFab, 2019-10-31:
The [diyHUE documentation](https://github.com/mcer12/Hugo-ESP8266/wiki/Sketch:-DiyHue) mentions in the "Registering the remote in Hue app" section : 

>  This will register the remote in diyHue and it will pop up in Hue app, in Accessories section.

It should be

>  This will register the remote in diyHue and it will pop up in the Philips Hue app, under **Settings > Accesory setup**, or under **Devices > Smart Controls** in the Hue Essentials app.

mcer12, 2019-10-31:
I gave you collaborator access, you should be able to edit the wiki now. Any contribution is appretiated just keep things simple ;)

closed 2019-10-31 by mcer12

MagicFab, 2019-10-31:
Thank you! Will do!