mcer12/Hugo-ESP8266 #17: MQTT_MAX_PACKET_SIZE error when compiling / sending MQTT firmware

MagicFab, 2019-11-01:
I am fairly new to Arduino.

I am using Debian 10 with Arduino IDE setup for ESP8255 module upload, I've added PubSubClient and EspMQTTClient libraires. I am trying to upload the latest MQTT firmware on a Hugo remote I recenly got.

I've connected the remote via USB, loaded the MQTT firmware files in Arduino IDE, when using "Verify/Compile", I get this error:
#error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to 512"

Uploading the diyHUE firmware works fine on the same remote.

Am I missing a particular library or requirement ? Thanks in advance.

mcer12, 2019-11-01:
Yes, the library packet size is too small by default and there's no other way to change it than in the library itself. You have to edit the library according to the error, that's why I put it there :D

MagicFab, 2019-11-01:
That worked, thank you!

closed 2019-11-01 by MagicFab