mcer12/Hugo-ESP8266 #22: Button press issue

user-hash, 2019-11-05:
Hey, just got Hugo today and I entered the config details normaly.
I got the mqtt working, but since I was just trying I didnt know which button is which, so I decided I will upload the code again, to add diffrent button topics and such.
Code uploads normaly, and I can enter the config screen and set the information, but when remote reboots, its only registering combo button presses, even if 1 button is pressed.
Any ideas how can I solve this?

Here is the debug code I get, if I press the buttons from 1-4 in slow sequence, no combo button presses. 
https://pastebin.com/wYSQja4F

mcer12, 2019-11-05:
Hi, this definitely shouldn't be happening. How fast are you pushing the buttons? You should make at least 1s delay between pushes, otherwise it can register it as a combo.

user-hash, 2019-11-05:
Please check the pastebin I posted above.
I press a single button 3 seconds after I get the "going to sleep" messege in the serial.
It only detects combo messeges..I can also record video, if you need it

mcer12, 2019-11-05:
In that case I suspect a hardware issue. This shouldn't be happening, please email me to investigate this further.

user-hash, 2019-11-05:
Hey, I hope you get this email, please confirm.
What information do you need from me?
Matic

closed 2019-11-05 by user-hash