mcer12/Hugo-ESP8266 #25: MQTT / Home Assistant / button topics

Zixim, 2019-12-05:
Hello,

just got 2 Hugos in the mail! I'm really chuffed, they feel like great little devices.

I flashed the MQTT firmware, made sure HA MQTT is configured for discovery (which it wasn't), and the Hugo device shows up under Integrations/MQTT in HA Configuration.
There's 2 entities , one for the status of the Hugo and one for the Battery.

I do have a couple of questions though:

Aren't all configured buttons supposed to show up in there ?
Why is the sensor for battery not showing any value ?

mcer12, 2019-12-05:
1) what do you mean by all buttons? Do you mean remotes or individual buttons on a single remote?
2) It only updates battery status together with a button topic. Maybe you didn't set up battery topic in web config?

Zixim, 2019-12-05:
I mean individual buttons, which I all set up in the web config as shown : 

[image]

The button presses show up in MQTT Lens , as expected : 
with topic `homeassistant/sensor/hugo_XXYYZZ/btn1` and payload `1`

The battery value never updates, and stays empty even after button presses.

mcer12, 2019-12-05:
You have to use example topics for all button topics as well as for battery topic for Home Assistant integration to work. See wiki:
https://github.com/mcer12/Hugo-ESP8266/wiki/Sketch:-MQTT
"homeassistant/sensor/hugo_[id]/state" for the buttons
"homeassistant/sensor/hugo_[id]/battery" for battery
I will add the note in config portal as well for more clarity in an upcoming update.

Zixim, 2019-12-05:
So the web config states : 
`Assign MQTT Topic and Payload for each button, for example:
Topic: "homeassistant/sensor/hugo_[id]/state"
Payload: "1" `

What does that mean I should set for Button 1 topic & payload ?
And what about Button 2?

mcer12, 2019-12-05:
Don't overthink it, it's meant to be simple :) 

B1
topic: homeassistant/sensor/hugo_[id]/state
payload: 1

B2
topic: homeassistant/sensor/hugo_[id]/state
payload: 2

etc......

Zixim, 2019-12-05:
[image]
Much better 😜

mcer12, 2019-12-05:
It's a common assumption that DIY products have to be difficult to set up, but I disagree :) Glad you managed to make it work! I did some edits to the wiki to make it a bit clearer.

closed 2019-12-05 by mcer12