mcer12/Hugo-ESP8266 #27: Wiki suggestion: a script to make bulk configuration easier

sowbug, 2020-03-02:
It's tedious to configure multiple remotes. I wrote a simple script that makes it a little easier.

```
REMOTES = {
    "12:23:34:45:56:67": 100,
    "12:23:34:45:56:68": 101,
    "12:23:34:45:56:69": 102,
}

PARAMS = {
    "ssid": "myssid",
    "pass": "mypass",
    "ip": "192.168.2.",
    "gw": "192.168.2.1",
    "sn": "255.255.255.0",
    "broker": "192.168.2.100",
    "port": "1883",
    "mqttusr": "mqttusername",
    "mqttpass": "mqttpass",
    "b1t": "homeassistant/sensor/hugo_[id]/state",
    "b1p": "payload1",
    "b2t": "homeassistant/sensor/hugo_[id]/state",
    "b2p": "payload2",
    "b3t": "homeassistant/sensor/hugo_[id]/state",
    "b3p": "payload3",
    "b4t": "homeassistant/sensor/hugo_[id]/state",
    "b4p": "payload4",
    "b5t": "homeassistant/sensor/hugo_[id]/state",
    "b5p": "payload5",
    "b6t": "homeassistant/sensor/hugo_[id]/state",
    "b6p": "payload6",
    "b7t": "homeassistant/sensor/hugo_[id]/state",
    "b7p": "payload7",
    "batt": "homeassistant/sensor/hugo_[id]/battery",
}

import copy
from urllib.parse import urlencode

for mac, ip in REMOTES.items():
    p = copy.deepcopy(PARAMS)
    p["ip"] += str(ip)
    print("%s: curl -d \"%s\" 10.10.10.1\n\n" % (mac, urlencode(p)))
```

This will spit out a bunch of `curl` commands you can execute on the command line when you're connected to a remote in configuration mode. The curl command is the same as filling out the web form and hitting submit. As you can see, my remotes are flashed with the MQTT firmware. Modify the script for other kinds.

mcer12, 2020-03-02:
Hi @sowbug I like the idea but I wonder how do you find out mac address before you connect to the network? Probably it can be found in debug console but that doesn't seem to simplify things. Can you elaborate on the process? 
What I do when I want to set up multiple remotes fast is connect to a remote in AP mode, save config and leave the config page open (with fields already filled in). Connect to another remote and just hit save. And repeat.

sowbug, 2020-03-03:
It's a two-pass operation. First, connect to each one and paste the MAC addresses into the script. Then start over again from the beginning and run the right generated curl script. Yeah, painful, I know.

I also started with the recycled web form, but once or twice I pressed submit a moment too early, before I'd gotten the DHCP assignment, and my browser wiped out all the fields. So I wanted something a little more predictable (including if I wanted to reflash the firmware in the future). Plus for various reasons it was helpful to have the list of MAC addresses when setting up the static addresses in my router, so once I compiled the list and noticed it looked a lot like a Python dict, the rest was quick.

To be clear, I'm not proud of this solution. But it did work!

Maybe a compromise would be an optional field on the form where you could paste JSON. You already have imported ArduinoJSON in the MQTT client, so it's not going to change the firmware size significantly.

mcer12, 2020-04-14:
Sorry to get to this so late, I will try to add a field for raw json in some reasonable way. 

Now that I think about it, maybe it would be best to make two more url paths
**/download_json**
**/import_json**
One with a textarea where you can paste the json and second that will just return raw json content that you can download. Root / will return the config website as usual

sowbug, 2020-04-14:
Cool, no hurry for me in particular -- mine are set up and working fine.

closed 2023-05-31 by mcer12