mcer12/Hugo-ESP8266 #30: Suggestion

StefanRehnberg, 2020-07-20:
Hi
It would be a very nice feature if you could implement "multi click" on the buttons, like the [Shelly i3](https://shelly.cloud/knowledge-base/devices/shelly-i3/).

[image]

mcer12, 2020-07-20:
Hi @StefanRehnberg this is sadly not possible. It's not possible because of HW reasons, in the way it works, it is impossible to distinguish between longpress and two quick presses. And even if it was possible, this would lead to worse battery performance since the remote would have to wait for possible second push every time. That said, I am planning on adding a long press (which is long overdue) which I think is going to give you plenty of options. At this point, there are single pushes and two-button combos, so 7 possible functions. With longpress it will go up  to 11 :)

StefanRehnberg, 2020-07-21:
Excellent! thanks for the info, keep up the good work :-)

closed 2020-07-21 by StefanRehnberg