mcer12/Hugo-ESP8266 #32: HTTP (Basic URL trigger)

patrickdej, 2020-09-11:
Hello,
we have inserted an URL into the Hugo, but the request is executed as a TCP message and not as an http request (seen in wireshark)
Whats wrong?

URL set to this http://172.20.20.153:8080/#/Main/Acknowledge?Sendkey=T1_S1_G
when executing this URL from an phone on the same network (just past it into the browser), it works fine.
so no issue on firewall.

thanks,

Patrick

mcer12, 2020-09-11:
Hi, I think this might have something to do with the port in your URL address. I never tried/needed this so this is probably what's the issue here. Lets start by removing the port from the URL address (so it will be 80 by default), see if it works.

patrickdej, 2020-09-18:
OK, works fine now. was due to the way the website worked, it needed more then only 1 url...
thanks!

closed 2020-10-25 by mcer12