mcer12/Hugo-ESP8266 #33: change firmware from URL to MQTT

patrickdej, 2021-01-08:
Hello,

We have been using the HUGO with the URL triggering and is working fine.
When we now want to change the firmware to the MQTT version, we get the following output, if we flash the URL version, all works fine. I'm quiet new to this, any direction where to go for finding a solution?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------#LF#CR#LF ets Jan  8 2013,rst cause:2, boot mode:(3,7)#CR#LF#CR#LFload 0x4010f000, len 3584, room 16 #CR#LFtail 0#CR#LFchksum 0xb0#CR#LFcsum 0xb0#CR#LFv2843a5ac#CR#LF~ld#LF#CR#LFFW: MqttTrigger#CR#LFButton: 0#CR#LF#LF--------------- CUT HERE FOR EXCEPTION DECODER ---------------#LF#CR#LFException (28):#CR#LFepc1=0x4020554a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000#CR#LF#CR#LF>>>stack>>>#CR#LF#CR#LFctx: cont#CR#LFsp: 3ffffd80 end: 3fffffc0 offset: 0190#CR#LF3fffff10:  3fffdad0 3fff0618 3fff06f8 40205539  #CR#LF3fffff20:  feefeffe feefeffe feefeffe feefeffe  #CR#LF3fffff30:  feefeffe feefeffe feefeffe feefeffe  #CR#LF3fffff40:  feefeffe feefeffe feefeffe feefeffe  #CR#LF3fffff50:  feefeffe feefeffe feefeffe feefeffe  #CR#LF3fffff60:  feefeffe feefeffe feefeffe feefeffe  #CR#LF3fffff70:  00000000 00000000 3fff185f 3fff185f  #CR#LF3fffff80:  00000000 feefeffe feefeffe feefeffe  #CR#LF3fffff90:  feefeffe feefeffe feefeffe 3fff0828  #CR#LF3fffffa0:  3fffdad0 00000000 3fff07e8 4020defc  #CR#LF3fffffb0:  feefeffe feefeffe 3ffe84f4 40100da9  #CR#LF<<<stack<<<#CR#LF#LF--------------- CUT HERE FOR EXCEPTION DECODER ---------------#LF#CR#LF ets Jan  8 2013,rst cause:2, boot mode:(3,7)#CR#LF#CR#LFload 0x4010f000, len 3584, room 16 #CR#LFtail 0#CR#LFchksum 0xb0#CR#LFcsum 0xb0#CR#LFv2843a5ac#CR#LF~ld#LF#CR#LFFW: MqttTrigger#CR#LFButton: 0#CR#LF#LF--------------- CUT HERE FOR EXCEPTION DECODER ---------------

regards,

patrick

mcer12, 2021-01-08:
I bet it's crashing because you don't erase whole flash :) So first change Erase flash to "all flash contents". In case I'm wrong, you might have some outdated libraries or ESP8266 board definition so check if you have the latest versions.

mcer12, 2021-01-08:
Also set frequency to 160MHz, the performance should be better.

mcer12, 2021-03-08:
Closing as stale, feel free to reopen.

closed 2021-03-08 by mcer12