mcer12/Hugo-ESP8266 #35: DHCP does not wait long enough

hagenbuch, 2021-01-30:
Dear Mcer12,

I am using 4 of your Hugos (firmware 1.4.1) on a regular basis and it works mostly fine, but -

While using it (over time) with two different Wifi routers from the same manufacturer (AVM, Fritzbox 7430 and 7530 - the last is very modern) the device seems to not wait long enough for the DHCP reply so it blinks instead of sending data. This happens every 1 in 5 times, no matter how distant I am.

I have seen that the timeout is hardcoded in your main code (4 and 8 seconds?) I am not proficient enough to write a pull request but you might figure this out in seconds.

I understand that you don't want to wait for too long because it affects battery life. However, I would really appreciate if this timeout could be configurable in the webfrontend or at least could be set as global variable in the code.

I hope that makes sense for you, too :-)

Thank you for your product and for taking your time..
Andreas

mcer12, 2021-01-30:
It should not take so long to receive DHCP. I would guess something with the network is the culprit. Have you tried static IP?

mcer12, 2021-03-08:
Closing as stale, feel free to reopen.

closed 2021-03-08 by mcer12