mcer12/Hugo-ESP8266 #37: not an issue, more a plea : Revive this project with esp32-c3 ?

Zixim, 2022-02-19:
The title says it all really...
There's this thing : https://github.com/makermoekoe/Picoclick-C3 , but it's far from user-friendly.
I've been using your Hugos for a few years now, and they're easy to config, run rock solid, integrate great with Home Assistant. No complaints at all.

So, a more modern esp32-c3 version, please ?

mcer12, 2022-02-20:
Hi, I don't see any issue with using ESP8266 - my hugos run prefectly fine and I don't see any meaningful benefit by using ESP32. If the ESP32 was 5GHz capable, that would be a different story.

That said, you can make the board yourself if you can solder, gerber, BOM and schematic are all in this repository :)

Zixim, 2022-02-20:
i'd prefer to purchase...8266 or 32 are equivalent in this usage case that's true.

closed 2022-02-20 by Zixim

ceoloide, 2022-02-21:
Shameless plug: a while ago I created an ESP-32 derivative of Hugo-ESP8622, aptly named [Ugo-ESP32](https://github.com/ceoloide/Ugo-ESP32).

That board uses a [TinyPICO](https://www.tinypico.com/) module, which uses almost no battery on low power mode. With a 1200mha battery I could get almost 2 years of standy time and over 500 button presses.

The ESP-32 gives more flexibility with encryption, so it supports both HTTPS and MQTTS endpoints at the same time.

JLCPCB now offers a wider variety of components for their SMD servive, so in theory one could order the Hugo-8266 through them directly and have it fully assembled.

Zixim, 2022-02-21:
This Tinypico thing is expensive - 22 usd ??

About JLCPCB + Hugo-ESP8266, I'm very tempted, but this is entirely new to me.
Even going through [the instructions included in Gerber_Hugo-ESP8266.zip](https://docs.easyeda.com/en/PCB/Order-PCB/), I'm left with way many questions, leading me to believe I'll never end up with a few useable devices.

I mean...it says : `1. Finish the schematic and PCB design at EasyEDA.`
But EasyEDA doesn't let me import the gerber files...

Slightly demotivated here...

mcer12, 2022-02-21:
You dont unzip the gerber díle, you upload it directly to jlcpcb ;)

Zixim, 2022-02-21:
Yea, that's what the web interface at JLCPCB told me... then it wants a CPL file, which I'm supposed to make in EasyEDA ?